RoboCup-Humanoid-TC/webots #1: Timing - Start of a game

MaikePaetzel, 2021-04-06:
Setup:

1. A valid game configuration is provided

Expected result:

1. All robots are spawned
2. For 2 minutes (real-time), the game state stays INITIAL
3. After 2 minutes (real-time), the game state changes to READY

omichel, 2021-04-07:
> 2. For 2 minutes (real-time), the game state stays INITIAL
> 3. After 2 minutes (real-time), the game state changes to READY

Why are these 2 minutes needed? This is not specified in the rules or any other documentation.
It actually takes about 16 seconds (real time) on my machine between the spawning of all the robots and GameController sending the READY state.

MaikePaetzel, 2021-04-07:
This is (newly) specified in Law 7 - Periods of play.

> Teams are given a 2 minute time period between the robot models being spawned and the start of the match.

Having such a period was requested by some teams to internally load machine learning models or other calibrate sensors if necessary.

medrimonia, 2021-04-10:
According to the test in #28, the real time spent in `INITIAL` is `120*real_time_factor` seconds, while it should not depend on `real_time_factor`. To keep this convenient for testing, I suggest adding a parameter `initial_state_duration` to `game.json` which would have a default to 120 and a very small value for all test scenarios.

medrimonia, 2021-04-10:
Covered by `test_initial_positions`.

omichel, 2021-04-12:
> According to the test in #28, the real time spent in INITIAL is 120*real_time_factor seconds, while it should not depend on real_time_factor. To keep this convenient for testing, I suggest adding a [...]

This is not the only duration that should be expressed in real time. So, if we want to improve this, we should add a more generic parameter to the `game.json` file that could be named `speed_up_real_time` and would actually speed up the real time waiting times as in the current version if true. If false (default), it will actually wait for the actual real time durations.

medrimonia, 2021-04-12:
I am not sure whether something was changed for this, but now at `real_time_factor=3`, it takes 6 minutes of simulated time (18 minutes real time) before going to ready.

omichel, 2021-04-13:
Sorry, it was broken when introducing the real time factor in the `game.json` file. Is now fixed in https://github.com/cyberbotics/webots/pull/2888/commits/2fa0d30facbbf2685483b9485b61047b5797de1e. It now takes 40 seconds of simulated time (2 minutes of real time) before going to ready.

medrimonia, 2021-04-20:
Covered automatically by the scenario `game_phases/initial`.

closed 2021-04-20 by medrimonia