RoboCup-Humanoid-TC/webots #14: Ball Being out of Play - Touch line, basic case

MaikePaetzel, 2021-04-06:
Setup:

1. Game is in state PLAY
2. Ball is touched by player of team red
3. The ball entirely crosses one of the touchlines

Expected outcome: 

1. ball is replaced on the centerline at the position it left the field; 
2. Game log contains the information (with real and simulated timestamp): position at which the ball left the field + robot last touching the ball (identified by team and number), position of the ball after being re-spawned on the field line

SoheilKhatibi, 2021-04-15:
> 1. ball is replaced on the centerline at the position it left the field;
Shouldn't it be ball is replaced on the **_touchline_** at the position it left the field? I think according to the Rules, line 645.

medrimonia, 2021-04-15:
Yes agreed, I updated the message.

medrimonia, 2021-04-20:
Manual test available in `ball_out/throw_in`.

closed 2021-04-25 by medrimonia