RoboCup-Humanoid-TC/webots #17: Ball Being out of Play - opponents' goal line, basic case

MaikePaetzel, 2021-04-06:
Setup:

1. Game is in state PLAY
2. Ball is touched by player of team red
3. The ball entirely crosses the goal line on the side of team blue

Expected outcome: 

1. ball is replaced on the centerline at the intersection with the centerline on the side the ball left the field
2. Game log contains the information (with real and simulated timestamp): position the ball left the field + robot last touching the ball (identified by team and number), position the ball was re-spawned

SoheilKhatibi, 2021-04-15:
> 1. ball is replaced on the centerline at the intersection with the centerline on the side the ball left the field

Shouldn't it be :  ball is replaced on the _**touchline**_ at the intersection with the centerline on the side the ball left the field.
I think according to the Rules, line 662.

While testing scenario #39 , the ball was respawned on the centerline, with its `Y` at the point it left the field, while it should be at the intersection of the touchline and centerline. (at the side it has left the field)

medrimonia, 2021-04-15:
Yes agreed and changed.

medrimonia, 2021-04-20:
Manual test available in `ball_out/goal_kick`.

closed 2021-04-25 by medrimonia