RoboCup-Humanoid-TC/webots #18: Ball Being out of Play - opponents' goal line, advanced case

MaikePaetzel, 2021-04-06:
Setup:

1. Game is in state PLAY
2. Ball is touched by player of team red
3. The ball entirely crosses the goal line on the side of team blue

Expected outcome: 

1. ball is replaced on the centerline at the intersection with the centerline on the side the ball left the field
2. AutoRef declares GOAL KICK team blue
3. Game log contains the information (with real and simulated timestamp): position the ball left the field + robot last touching the ball (identified by team and number), position the ball was re-spawned, goal kick and team it was awarded to

omichel, 2021-04-14:
> 1. ball is replaced on the centerline at the intersection with the centerline on the side the ball left the field

I would assume this instead (according to the rule book):

1. ball is replaced on the touchline at the intersection with the centerline on the side the ball left the field

medrimonia, 2021-04-15:
Yes, agreed and fixed.