RoboCup-Humanoid-TC/webots #21: Detection and Penalty of Fallen Robots - single robot

MaikePaetzel, 2021-04-06:
Setup:

1. Game is in state PLAY
2. Robot falls down (i.e., not only feet are touching the floor)
3. Robot does not get up for 20 seconds

Expected outcome: 

1. Robot receives a 30-second removal penalty
2. Robot is respawned at a position outside the touchline which is further away from the ball in its own half at the height of the penalty mark
3. Game log contains the information (with real and simulated timestamp): robot that was penalized (identified by team name and number) + offense that was called

omichel, 2021-04-09:
> 2. Robot is respawned at a position outside the touchline which is further away from the ball in its own half at the height of the penalty mark

How far away from the touchline? like 50 cm?

omichel, 2021-04-09:
Note: we need to modify the Darwin-OP proto and rename the feet with the [feet] prefix in order to be able to implement this.

medrimonia, 2021-04-09:
> How far away from the touchline? like 50 cm?

According to `reentryStartingPose` from the `game.json` file, with a possible change of sign on the `y` and rotation of pi around 'z' axis depending on the side of the touch line (and a possible change of sign of `x` depending on the robot team's).

> Note: we need to modify the Darwin-OP proto and rename the feet with the [feet] prefix in order to be able to implement this.

Do you want us to do it?

omichel, 2021-04-09:
> Do you want us to do it?

No, I can do it, it should be quick.

medrimonia, 2021-04-20:
Manual test available in `misc_fouls/robot_falling`.