RoboCup-Humanoid-TC/webots #22: Detection and Penalty of Fallen Robots - multiple robots

MaikePaetzel, 2021-04-06:
Setup:

1. Game is in state PLAY
2. Robot 1 of team red falls down (i.e., not only feet are touching the floor)
3. Robot 2 of team red falls down (i.e., not only feet are touching the floor)
4. Robot 1 does not get up for 20 seconds
5. Robot 2 does not get up for 20 seconds

Expected outcome: 

1. Robot 1 receives a 30-second removal penalty
2. Robot 1 is respawned at a position outside the touchline which is further away from the ball in its own half at the height of the penalty mark
3. Robot 2 receives a 30-second removal penalty
4. Robot 2 is respawned at a position outside the touchline which is further away from the ball in its own half 60 cm away from the other robot that is already suffering from a removal penalty
3. Game log contains the information (with real and simulated timestamp): both robots that were penalized (identified by team name and number) + offense that was called, penalty position they were respawned at

omichel, 2021-04-09:
> 60 cm away from the other robot that is already suffering from a removal penalty

Could you be more precise on that? Is it 60 cm towards to goal line, towards the center line or towards the outside of the field?

medrimonia, 2021-04-09:
It's L858-866 of [this document](https://cdn.robocup.org/hl/wp/2021/04/V-HL21_Rules_Draft3.pdf) for the exact rule.

In this case, both solutions are acceptable, so implementation is up to you (the logical way with respect to how rules are written would be to take the furthest one from the ball).

medrimonia, 2021-04-20:
Manual test available in `misc_fouls/robot_falling`.