RoboCup-Humanoid-TC/webots #3: Robot Positioning - Illegal position (1)

MaikePaetzel, 2021-04-06:
Setup:

1. In the config.json, the player is positioned outside of the field lines with parts of the feet touching the field line

Expected outcome:  

1. The robot is spawned at the beginning of the game. 
2. Once the state is changed to READY, the robot receives an immediate 30-second removal penalty.

omichel, 2021-04-07:
In case a robot is at an invalid position, I assume the referee has to:
1. send a PENALTY:team:player_id:INCAPABLE to the GameController.
2. move the robot at location `reentryStartingPose` specified in the `team.json` file for that robot.
3. "lock" the robot at this location for the next 30 seconds: reset it to this location in case the robot is attempting to move away.

Is that correct?

MaikePaetzel, 2021-04-07:
Step 1 and 2 is correct, but it doesn't need to lock the robot at its penalty position. Robots are required to listen to the GameController to understand whether they are penalized. If they try to move while in a penalty, they are re-penalized for 30 seconds. If they don't listen to the GameController, this may mean that they remain penalized for the rest of the game because they get frequent re-penalizations, but this is a feature, not a bug in the rules.

SoheilKhatibi, 2021-04-07:
I think in this test, all the robots' feet will touch the lines with changing the "halfTimeStartingPose" translation from [-3.5, -3.06, 0.24] to something like [-3.5, -3.05, 0.24] or [-3.5, -3, 0.24]. @omichel I think if you detect being on the touchline by projecting the convex hull of the robot, they are even in an illegal position with their default translation because the projection of their knee is on the lines.
So I think it can be tested now.

MaikePaetzel, 2021-04-07:
Can you please provide, test, and upload a configuration file for which this is the case?

SoheilKhatibi, 2021-04-07:
[Illegal position_1.zip](https://github.com/RoboCup-Humanoid-TC/webots/files/6274124/Illegal.position_1.zip)
This could be on of the cases

omichel, 2021-04-08:
@SoheilKhatibi: Thank you, it works as expected. From the log:

> PENALTY:1:1:INCAPABLE

medrimonia, 2021-04-10:
Covered by `test_initial_positions`.

medrimonia, 2021-04-12:
In the latest version, robots are penalized while state is still `INITIAL`, thus they have already exited penalty when state `READY` is opened.

omichel, 2021-04-13:
Sorry, this was broken by the implementation of new features. It is now fixed in this commit: https://github.com/cyberbotics/webots/pull/2888/commits/9baa4b10431fdf60b05118405bf88732c1c97ce1

medrimonia, 2021-04-20:
Covered automatically by the scenario `positioning/initial`.

closed 2021-04-20 by medrimonia