RoboCup-Humanoid-TC/webots #4: Robot Positioning - Illegal position (2)

MaikePaetzel, 2021-04-06:
Setup:

1. In the config.json, the player is positioned entirely outside the field with one leg in the left half of the field and one leg in the right half of the field

Expected outcome:  

1. The robot is spawned at the beginning of the game. Once the state is changed to READY, the robot receives an immediate 30-second removal penalty.

SoheilKhatibi, 2021-04-07:
setting "halfTimeStartingPose" translation of one robot to [0, -3.20, 0.24] will put it in that situation. It can also be tested.

MaikePaetzel, 2021-04-07:
Can you please provide, test, and upload a configuration file for which this is the case?

SoheilKhatibi, 2021-04-07:
[Illegal position_2.zip](https://github.com/RoboCup-Humanoid-TC/webots/files/6274150/Illegal.position_2.zip)

omichel, 2021-04-08:
The test passes. From the log:

> INCAPABLE penalty for red player 1: halfTimeStartingPose outside team side

medrimonia, 2021-04-12:
In the latest version, robots are penalized while state is still `INITIAL`, thus they have already exited penalty when state is changed to `READY`.

omichel, 2021-04-13:
Fixed in https://github.com/cyberbotics/webots/pull/2888/commits/9baa4b10431fdf60b05118405bf88732c1c97ce1.

medrimonia, 2021-04-24:
This test is now parts of the automated test `positioning/initial`.

closed 2021-04-24 by medrimonia