RoboCup-Humanoid-TC/webots #5: Game Setup - first half time

MaikePaetzel, 2021-04-06:
Setup:

1. In the game config file, two teams are declared for a normal KidSize match, one team with 4 players, one with 3

Expected outcome: 

1. The respective number of robots is spawned at the beginning of the game. 
2. The teams randomly receive a side they play on. 
3. Kick-off is randomly set to a team (i.e., if the same test case is run multiple times, the side and kick-off of teams differ)
4. Game state switches to READY
5. After 45 seconds game state switches to SET
6. The ball is placed on the center mark
7. After 15 seconds, the game state switches to PLAY

omichel, 2021-04-07:
> 7. After 15 seconds, the game state switches to PLAY

The rule book (draft 3) says (line 810): "The referee gives the signal 'PLAY' 5 seconds after the 'SET' signal was given."

Which one do you want to implement, 5 or 15 seconds?

MaikePaetzel, 2021-04-07:
Sorry my bad, I confused the timing with the free kick! I've changed it in the original issue.