RoboCup-Humanoid-TC/webots #6: Game Setup - second half time

MaikePaetzel, 2021-04-06:
Setup:

1. In the game config file, two teams are declared for a normal KidSize match, one team with 4 players, one with 3
2. 10 minutes of game time in the first half elapses

Expected outcome: 

1. The AutoRef calls the first half time to be over
2. The robots respawn on the starting positions at the side of the field
3. The sides of play and the kick-off have flipped in the GameController
4. 15 seconds (real-time) elapse
5. The game state switches to READY
6. After 45 seconds game state switches to SET
7. The ball is placed on the center mark
8. After 15 seconds, the game state switches to PLAY

omichel, 2021-04-08:
I send the "FINISH" message to GameController and it seems to work. However, when I send the "READY" message, I am getting an illegal answer from GameController with this READY message. From the GUI, it seems I should press the "2nd Half" button, but there is no such message in the TCP/IP protocol. Is it missing? Or did I miss something?

MaikePaetzel, 2021-04-08:
Sorry I missed that part! I thought the GC automatically switches to the second half, but it seems it doesn't (or at least not anymore with setting the half-time break to 0 seconds). I've pushed a fix to this branch: https://github.com/RoboCup-Humanoid-TC/GameController/tree/advanced-autoref-updates
You can now send STATE:SECOND-HALF to explicitly switch to the second half.

medrimonia, 2021-05-20:
Automatically tested and passing.

closed 2021-05-20 by medrimonia