RoboCup-Humanoid-TC/webots #7: Detection of a Goal Being Scored - Simple goal

MaikePaetzel, 2021-04-06:
Setup:

1. Player of team red in front of blue’s goal
2. Player kicks ball entirely over the goal line

Expected outcome: 

1. Goal scored for team red
2. Game state switches to READY 
3. After 45 seconds, the game state switches to SET
4. The ball is spawned at the center mark with team blue having kick-off

SoheilKhatibi, 2021-04-14:
While testing scenario #38, after scoring a goal, the ball is spawned at the center mark after the state changes to `READY`. According to the rule book (draft 3, line 1679): The ball is placed stationary on the centre mark after the `SET` signal was given.
so it should be spawned at the center mark after the state changes to `SET`. I think it is also true for starting the first and second half. But I don't exactly know where the ball should be at the beginning of first and second half.

medrimonia, 2021-04-24:
This test is now automated and successfully passing  in `scoring/goal`.

closed 2021-04-24 by medrimonia