RoboCup-Humanoid-TC/webots #24: Handling of extra time?

omichel, 2021-04-07:
For each goal scored, it takes about one minute (READY & SET states) to resume the actual play, but during this time, the game controller doesn't stop the 10 minutes timer. For kicks (penalty/direct/indirect/side), there is also some significant amount of time during which players have to move away from the ball. Again, this time is not played, but accounted by the game controller. At the end of a 10 minutes half, several minutes are likely to be spent not playing, but walking to a legal position (in READY & SET states).

So my question is: shall the referee account for non-played time and when the official 10 minutes time is out (e.g., remaining seconds is 0), it should wait for the accumulated extra time to elapse before it send the FINISH message?

MaikePaetzel, 2021-04-07:
No, whatever the GameController counts as in-play time is counted as in-play on purpose. The only reasons for extending a half-time period is (a) if there was major unexpected interference in a game (which I do not foresee happening in the simulated competition as there are no humans involved in refereeing and/or handling the robots) or (b) to grant robots a chance to finish their final attempt at scoring a goal (which is difficult for the AutoRef to detect, I assume).

omichel, 2021-04-07:
Thank you for the explanation.

closed 2021-04-07 by omichel