RoboCup-Humanoid-TC/webots #26: Robot positioning - Illegal position during READY

MaikePaetzel, 2021-04-08:
Setup:

1. In the config.json, the player is positioned in a valid position outside the field of play
2. The robot walks into the field and comes to stand in a valid position during the READY phase
3. During the set phase, the robot falls into the opponent's half of the field

Expected outcome:

1. The robot is spawned at the beginning of the game.
2. The game state changes to READY
3. The game state changes to SET
4. The robot is penalized as soon as the robot touches the opponent's half of the field
5. The robot is respawned at the penalty position
6. The game state changes to PLAY
7. The penalty is removed from the robot again
5. Game log contains the information (with real and simulated timestamp): Robots being spawned + position,  game state READY, game state SET, robot (identified by team and number) penalized + type of penalty, robot (identified by team and number) re-spawned at penalty position, game state PLAY, robot (identified by team and number) penalty ended

medrimonia, 2021-04-12:
On the latest version, I checked it by moving four robots manually to the opponent's field during the set phase, but none of them was penalized. Once an automated scenario is prepared, we will be able to have a reproducible case.

SoheilKhatibi, 2021-04-14:
While testing this issue with this scenario #37 , the robot `Blue 1` who is manually fallen into the opponent's field was not penalized.
[image]

medrimonia, 2021-04-20:
Manual test available in `positioning/ready`.

medrimonia, 2021-04-24:
Now test is automated, but when I run it, this case is still failing.

medrimonia, 2021-05-20:
All tests are now passing.

closed 2021-05-20 by medrimonia