RoboCup-Humanoid-TC/webots #31: Robot positioning - Illegal position in the end of READY (3)

medrimonia, 2021-04-09:
Setup:

1. In the config.json, the player is positioned in a valid position outside the field of play
2. The robot from does not move before the end of the READY phase, and is therefore outside of the field when state changes to SET.

Expected outcome:

1. The robot is spawned at the beginning of the game.
2. The game state changes to READY
3. The game state changes to SET
4. The robot is penalized as soon as the state changes to SET
5. The robot is respawned at the penalty position
6. The game state changes to PLAY
7. The penalty is removed from the robot again
5. Game log contains the information (with real and simulated timestamp): Robots being spawned + position,  game state READY, game state SET, robot (identified by team and number) penalized + type of penalty, robot (identified by team and number) re-spawned at penalty position, game state PLAY, robot (identified by team and number) penalty ended

medrimonia, 2021-04-10:
Covered by `test_initial_positions`.

medrimonia, 2021-04-12:
When starting the latest version with the default configuration and just letting time pass, robots are not moving at all but they are not being penalized.

omichel, 2021-04-13:
>When starting the latest version with the default configuration and just letting time pass, robots are not moving at all but they are not being penalized.

Fixed in https://github.com/cyberbotics/webots/pull/2888/commits/d5d9c3d2fc1b802cdc1d7d5bd30072468a23682c

medrimonia, 2021-04-20:
Covered automatically by the scenario `positioning/initial`.

medrimonia, 2021-04-20:
According to the automatic test, robots are getting penalized at the beginning of PLAYING and not at the beginning of SET (which is important to guarantee that they are allowed to move directly from the beginning of PLAYING).

omichel, 2021-04-20:
> According to the automatic test, robots are getting penalized at the beginning of PLAYING and not at the beginning of SET (which is important to guarantee that they are allowed to move directly from the beginning of PLAYING).

Fixed in https://github.com/cyberbotics/webots/pull/2888/commits/56dd14c6b6db6784abd0728848ccc1e77e73324d.

closed 2021-04-20 by medrimonia