RoboCup-Humanoid-TC/webots #32: Updating sensor timesteps duplicate the sensor

medrimonia, 2021-04-10:
While trying to use the default `client` provided and looking at the network and messages, I noticed that the size of the messages sent by the `player`was growing.

After investigating, I noticed that each request to update the sensor timesteps add another sensor because `robot->getDevice()` creates a new object every time. A quick test is replacing line  `sensors.insert(device);` by the following:
```
                  printf("adding sensor '%s' with address %p\n", sensorTimeStep.name().c_str(), device);
                  sensors.insert(device);
                  printf("New number of sensors: %ld\n", sensors.size());
```

Which outputs something like (I dropped irrelevant lines):
```
adding sensor 'NeckS' with address 0x5613b24308c0
New number of sensors: 1
Sending a message of size 2 (with 0 positions)
...
adding sensor 'NeckS' with address 0x5613b24419d0
New number of sensors: 2
Sending a message of size 20 (with 1 positions)
...
adding sensor 'NeckS' with address 0x5613b2454ae0
New number of sensors: 3
Sending a message of size 38 (with 2 positions)
...
adding sensor 'NeckS' with address 0x5613b2457960
New number of sensors: 4
Sending a message of size 56 (with 3 positions)
```

A simple suggestion would be to use a `std::map<std::string, webots::Device *>` with the name of the sensor rather than a set.

omichel, 2021-04-12:
I appears to be a bug in Webots as the `robot->getDevice()` method should return the same value for the same argument. I am fixing it now.

omichel, 2021-04-12:
It should be fixed now in https://github.com/cyberbotics/webots/pull/2957.

closed 2021-04-12 by omichel