RoboCup-Humanoid-TC/webots #34: Illegal field entry - yellow card

medrimonia, 2021-04-10:
Setup:

1. In the config.json, the robot is positioned in an invalid position inside the field of play.
2. The robot walks into the field at the beginning of the READY phase, while it is still penalized.

Expected outcome:

1. The robot is spawned at the beginning of the game.
2. The game state changes to READY
3. The robot is penalized by the referee and respawned at penalty position
4. The robot enters the field again
5. The robot is penalized again by the referee, respawned at penalty position and shown a yellow card
6. Game log contains the information (with real and simulated timestamp): Robots being spawned + position,  game state READY, robot (identified by team and number) penalized + type of penalty, robot (identified by team and number) re-spawned at penalty position, robot (identified by team and number) penalized again, robot (identified by team and number), robot (identified by team and number) re-spawned at penalty position receiving a yellow card

medrimonia, 2021-04-10:
Covered by `test_initial_positions`.

omichel, 2021-04-14:
I would assume this is a more general rule: if a penalized robot enters the field again, it gets a yellow card. Is that correct?
I could not find it explicitly in the rule book.

medrimonia, 2021-04-15:
Yes, that's the case, it's written L525 of the [Rules-Draft3](https://cdn.robocup.org/hl/wp/2021/04/V-HL21_Rules_Draft3.pdf)

medrimonia, 2021-04-20:
Covered automatically by the scenario `positioning/initial`.

medrimonia, 2021-04-20:
According to the automatic test, robots are getting penalized at the beginning of PLAYING and not at the beginning of SET (which is important to guarantee that they are allowed to move directly from the beginning of PLAYING).

omichel, 2021-04-20:
> According to the automatic test, robots are getting penalized at the beginning of PLAYING and not at the beginning of SET (which is important to guarantee that they are allowed to move directly from the beginning of PLAYING).

Fixed in https://github.com/cyberbotics/webots/pull/2888/commits/56dd14c6b6db6784abd0728848ccc1e77e73324d

closed 2021-04-20 by medrimonia