RoboCup-Humanoid-TC/webots #35: Symetry for poses provided by team is only valid for specific values

medrimonia, 2021-04-10:
The current implementation of the symetry in the `referee.py` file shown below is only valid if the axis of the rotation is the `[0 0 1]` vector.

`halfTimeStartingRotation[3] = math.pi - halfTimeStartingRotation[3]`

SoheilKhatibi, 2021-04-11:
I was faced with a similar problem before. One possible solution could be to convert axis-angle rotation to euler angles. then to apply that symmetry, just do that (`angle = math.pi - angle`) with the rotation around the Z- axis. and again convert those euler angles to axis-angle rotation. see [https://www.euclideanspace.com/maths/geometry/rotations/conversions/angleToEuler/index.htm](this) and [https://www.euclideanspace.com/maths/geometry/rotations/conversions/eulerToAngle/](this).
I tried to implement that and wanted to create a pull request. but I faced with something really strange. The process seems to work okay but the robot surprisingly starts to fall apart. the code is on symmetryProblem branch of [https://github.com/SoheilKhatibi/webots_tests/tree/symmetryProblem](this) project.
[image]

omichel, 2021-04-12:
I just fixed this by converting the axis angle to quaternion, perform the rotation along the Z axis and convert back to axis angle. Note: I had to include the transforms3d pip dependency to facilitate the quaternion operations. See: https://github.com/cyberbotics/webots/blob/f0d026ea2943dfccf63835307d420273018448b1/projects/samples/contests/robocup/controllers/referee/referee.py#L302-L308

closed 2021-04-12 by omichel

omichel, 2021-04-13:
Well, this problem is actually not fixed in the current version, but I am reluctant to fix it because it would encourage competitors to design robot protos that don't respect our design rules for [orientation](https://cyberbotics.com/doc/reference/proto-design-guidelines#orientation) and [translation and rotation](https://cyberbotics.com/doc/reference/proto-design-guidelines#translation-and-rotation). At initial position, the robots should be in their default stand-up position, with possibly a rotation along the Z axis (to look towards their goal or the opponent's goal or backwards if they want to), but the rotation axis should not be changed otherwise the robot would not be in a stable position.