RoboCup-Humanoid-TC/webots #41: Image sent by player.cpp might be incomplete

medrimonia, 2021-04-15:
In player.cpp, the lines sending the image might end up sending incomplete images when the image buffer contains a 0 value.

```
                CameraMeasurement *measurement = sensorMeasurements.add_cameras();
                measurement->set_name(camera->getName());
                measurement->set_width(camera->getWidth());
                measurement->set_height(camera->getHeight());
                measurement->set_quality(-1);  // raw image (JPEG compression not yet supported)
                measurement->set_image((const char *)camera->getImage());
```
Should be changed to
```
                CameraMeasurement *measurement = sensorMeasurements.add_cameras();
                int img_width = camera->getWidth();
                int img_height = camera->getHeight();
                measurement->set_name(camera->getName());
                measurement->set_width(img_width);
                measurement->set_height(img_height);
                measurement->set_quality(-1);  // raw image (JPEG compression not yet supported)
                measurement->set_image(camera->getImage(), img_width*img_height*3);
```

omichel, 2021-04-16:
This is not as simple as this as the image returned by a Webots camera has RGBA format (32 bit per pixel).
So, we should convert it to RGB before sending it to the controller.

omichel, 2021-04-16:
Fixed in https://github.com/cyberbotics/webots/pull/2888/commits/f9ed79f3279c619c918b6ecb23b22e30f2d893d8