RoboCup-Humanoid-TC/webots #53: Forceful contact - Contact in goal area

medrimonia, 2021-04-18:
This situation include 3 variations: a,b,c, when they differ, it is explicitely
specified.

## Setup:

- GameState is READY and ball is in play
- Robot G is goalkeeper and is in its own goal area
- Robot F is from the oppsite team and is
  - a: outside the goal area of robot G but close to it
  - b+c: inside the goal area of robot G
- The ball is placed:
  - a+b: Less than 1 meter from the robot
  - c: More than 1 meter away from the robots
  
## Procedure:

- a. A contact occurs between the two robots
- b. G moves at speed > 0.2 m/s towards F who is static
- c. G and F move toward each other at speed < 0.2 m/s

## Expected outcome

- Robot G is not penalized
- a+b. A free kick in favor of the team of robot G is called, F is not penalized
- c. Robot F is penalized

omichel, 2021-04-19:
Shall we assume that the goal keeper is player number 1?

omichel, 2021-04-19:
> b. G moves at speed > 0.2 m/s towards F who is static

What does static mean?
Can we consider a robot static if the velocity of its main body is close to zero (below some threshold, like 0.01 m/s).

medrimonia, 2021-04-19:
> Shall we assume that the goal keeper is player number 1?
Players have to be able to change the goal keeper during so this is at least something that will need to change. However, I think that the feature for informing the referee that the goalie has changed is not available yet. Moreover I don't think this information is sent through the UDP. So I think that there will be need of some changes in the communication to implement this. @MaikePaetzel can you confirm this?

> Can we consider a robot static if the velocity of its main body is close to zero (below some threshold, like 0.01 m/s).

Yes, that's the intent. The *explicit* definition is less than `s_t =0.2 [m/s]`, but for pushing scenario I think it's better to take really a minimal speed.

omichel, 2021-04-19:
> Yes, that's the intent. The explicit definition is less than s_t =0.2 [m/s], but for pushing scenario I think it's better to take really a minimal speed.

So, which value shall we use for that?

medrimonia, 2021-04-19:
I updated using 0.05 m/s.