RoboCup-Humanoid-TC/webots #54: Ball holding - Multi-robot, KidSize

medrimonia, 2021-04-18:
This situation include 4 variations: a,b,c when they differ, it is explicitely
specified.

## Setup:

- GameState is READY, ball is in play and the game is a KidSize game
- R1 is at less than 30 centimeters from the ball
- R2 is exactly on the other side of the ball
  - a+b. 30 centimeters away from the ball
  - c. 1 meters away from the ball
- a+c. R1 and R2 are from the same team
- b. R1 and R2 are from opposite team
  
## Procedure:

- 1 seconds elapses with all robots static

## Expected outcome

- a. A free kick in favor of the opposite team is given, no robot is penalized.
- b+c. Free kick is not called, no robot is penalized.

medrimonia, 2021-04-20:
Automated test is available in `ball_holding/multi_robot_kid`.

omichel, 2021-04-20:
When a robot holds the ball, a free kick is awarded to the opponent team, but happen to the faulty robot? Does it get a `BALL_MANIPULATION` penalty and is removed from the game?

medrimonia, 2021-04-20:
The faulty robot(s) are not penalized, they have to move autonomously out from the ball to proceed with the free kick.

medrimonia, 2021-05-20:
All tests are now passing.

closed 2021-05-20 by medrimonia