RoboCup-Humanoid-TC/webots #56: Logging of information

medrimonia, 2021-04-19:
Issue based on the conversation of #51 

> The time information appears in the log file, but not in the Webots console. The motivation is that the output in the Webots console is usually checked live and hence the timing information is not useful there (and makes some extra noise), whereas it is important in the log.txt file.

1. This log file is systematically empty for me so I have not been able to compare to it. It is probably because it is the `test_supervisor` that quits the simulation (game_controller is not closed either).

2. I think that it's problematic to have a hard coded file here because if we launch two tests on a row, we're destroying the logs (except if the caller takes the explicit responsibility of moving the file to another folder before launching the second test)

3. I think all of this would be made simpler if the standard output is used rather than a log file. Would this be fine for you @MaikePaetzel ? It would also solve the fact that `test_supervisor` and `referee` cannot access the same file, which means that we have to cross the data from both files when debugging.

omichel, 2021-04-19:
1. Normally, the referee should be able to quit cleanly, including closing the GameController and the `log.txt` file when you request the quit from the test supervisor. I will investigate why this doesn't seem to work and hopefully fix it.
2. We can add an environment variable here to define the log file, like `WEBOTS_ROBOCUP_LOG=/path/to/my/log.txt`.
3. It is indeed possible to redirect the stdout/stderr of Webots to a file with the `--stderr` and `--stdout` [command line options](https://cyberbotics.com/doc/guide/starting-webots#command-line-arguments). However, that means that not only the stdout/stderr of the referee will be redirected there, but also some Webots messages, like warnings, errors, infos, starting of controllers, messages from other controllers, etc. which won't have the same format as the `log.txt` file, including the real/simulation timestamp. Moreover, it turned out to be very convenient to me to see the messages popping in the Webots console while developing and running the manual tests. So, I believe, it is better to keep the log file in a regular file. We may flush it regularly to ensure it is always up-to-date.

omichel, 2021-04-21:
> 1. This log file is systematically empty for me so I have not been able to compare to it. It is probably because it is the test_supervisor that quits the simulation (game_controller is not closed either).

I could reproduce and just fixed this problem. Can you check again and confirm the log file is not empty any more?

medrimonia, 2021-04-21:
The log is now properly filled which helped a lot for debug, thanks!

closed 2021-04-28 by omichel