RoboCup-Humanoid-TC/webots #58: Forceful contact - Pushing

medrimonia, 2021-04-19:
This situation include 4 variations: A,B,C,D. When they differ, it is explicitely
specified.

- A. Back-pushing
- B. Side-pushing
- C. Pushing occurs in middle range of the ball -> penalty
- D. Pushing occurs very far from the ball -> nothing happens

## Setup:

- GameState is READY and ball is in play
- Robots R1 and R2 are nearby, they are of opposite team, not penalized and
  outside any penalty area.
- The ball is placed:
  - A. At less than 0.5 meters from R1 and more than 0.6 meters from R2
  - B. At a similar distance of R1 and R2, closer than 1 meter
  - C. At less than 1.5 meters of R2 and more than 1.6 meters of R1
  - D. More than 2 meters away from the closest robot 
  
## Procedure:

- Both robots enter in contact repeatedly at very low speed for more than 2 seconds

## Expected outcome

- A. Direct free-kick is awarded to the team of R1
- C. No freekick, but R1 is penalized
- B+D. Nothing happens (no penalty, no FreeKick)

omichel, 2021-04-19:
> Both robots enter in contact repeatedly at very low speed for more than 2 seconds

Can you specify a value for this "very low speed"? I would assume it corresponds to the linear velocity of the main body of the robot.

omichel, 2021-04-19:
> B. At a similar distance of R1 and R2, closer than 1 meter

What do you mean precisely with "similar"? What is a amount of the tolerance? Can you provide a mathematical formulae?

medrimonia, 2021-04-19:
Updated for both. The issues mention scenario and should therefore not be used inside code. For the exact value to be used for all the forceful contact rules, it should be based on Figure 4 and Table 2 of the rule document.

medrimonia, 2021-05-20:
Note for the future: this scenario doesn't seem to be tested.