RoboCup-Humanoid-TC/webots #60: Convex hull computation simplification

omichel, 2021-04-20:
I have been investigating the convex hull path and it is very difficult to achieve within a reasonable performance. Basically we should first retrieve a list of all the solids composing each robot and then iterate repeatedly:

1. retrieve the position of every solid and project the points on the ground for each robot.
2. compute the AABB of these points for each robot (to optimize collision detection, broad phase).
3. compute the convex hull of these points.
4. compute the Delaunay triangulation for the convex hull.
5. perform many triangle circle collision detections.

This is badly affecting the performance and we cannot respect the 1/3 real time constraint with that.
Even if we perform this computation not every time step, this is still very complicated and heavy computation.

A solution to simplify this process would be to consider only the 2 feet and the 2 hands instead of all the solid nodes of the robot.

medrimonia, 2021-04-20:
If steps 3 to 5 are too complex, I am concerned that if three or four robots of the same team are in the vicinity of the ball, we will end up with the same problem (convex hull will be of 12-16 points). Would it also be a problem?

One of the major issue with moving only with two feet and two hands is that it can form a line rather than a polygon. So even if the ball is between the two feet of the robot, the robot might not be committing a foul depending on where his hands are.

omichel, 2021-04-20:
> If steps 3 to 5 are too complex, I am concerned that if three or four robots of the same team are in the vicinity of the ball, we will end up with the same problem (convex hull will be of 12-16 points). Would it also be a problem?

Probably. But the number of points should be nevertheless much lower than with the original version (based on the center of every solid of each robot), but still... A simplification may consist in projecting the center of mass of the robots on the floor and computing the collision between the ball and and each robot convex hull, plus between the ball and:
1. the segment between two robots, if two robots are in the vicinity of the ball.
2. the triangle between three robots, if three robots are in the vicinity of the ball, and
3. the quad between the four robots, if four robots are in the vicinity of the ball.

> One of the major issue with moving only with two feet and two hands is that it can form a line rather than a polygon. So even if the ball is between the two feet of the robot, the robot might not be committing a foul depending on where his hands are.

This is also the case in the original version (based on the center of every solid of each robot). Since we now compute the collision between the ball and the simple convex hull, in the rare case of a line, there would still be a collision if the ball is between the two feet of a robot, and hence a foul, regardless the position of the hands.

omichel, 2021-04-20:
I just completed a first implementation of the quad convex hull (two feet + two hands) for detecting ball holding for a single robot and the performance impact although not negligible seems acceptable.

closed 2021-05-20 by omichel