RoboCup-Humanoid-TC/webots #104: Merging the develop branch from Webots broke the Player

medrimonia, 2021-05-11:
When a client connects and try to send some messages, the player now outputs a message: `Error: wb_compass_enable(): invalid device tag.` While the same client was previously working perfectly fine.

I checked more in details to find which commit introduced the bug and found the following:

- On e4b12c0 : the client runs without any error
- On a66fac0 : the client fails with the previously mentioned error

It is to be noted that neither the PROTO used nor the Player reference compass at any moment.

medrimonia, 2021-05-11:
Yes I did, I even ran a `git clean -xfd` after switching branch.

omichel, 2021-05-12:
I just tested it with the provided sample [client](https://github.com/RoboCup-Humanoid-TC/webots/blob/experimental/projects/samples/contests/robocup/controllers/player/client.cpp) and it works nicely without an error message.

omichel, 2021-05-12:
On Linux, I am getting a very strange error when trying to execute the sample client:
```
terminate called after throwing an instance of 'std::system_error'
  what():  Unknown error -1
Aborted (core dumped)
```
This error disappears if I modify client.cpp (comment out the protobuf code) and don't link it with messages.pb.cpp. @medrimonia : do you have any idea about it?

medrimonia, 2021-05-12:
Yep, I faced it on a personal client and it was related to linking with `protobuf` without linking to `lpthread` library (order of library matters). I was not sure if that was something related to my local configuration, I solved it on another client.

Last time I tried to fix it using the `client` in there, I had no satisfying results quickly, I will try to get a look at it if I'm not the only one experiencing it.

medrimonia, 2021-05-12:
Just replacing in `Makefile.client` the line:

```LIBS += -lprotobuf```

With

```LIBS += -lprotobuf -lpthread```

Was sufficient for me, but I guess that this can cause issues on windows.

omichel, 2021-05-12:
Thank you. It solved the problem indeed.
We can make a Windows specific case to avoid any conflict on Windows.

medrimonia, 2021-05-12:
I was able to test the default client with the latest version of experimental and I can confirm that it works. I'll have to get into more details with my other client to produce a stronger basis for you to reproduce the bug.

omichel, 2021-05-12:
OK, so I was able to test the current `experimental` branch on both Linux and Windows with the sample `client` and I could not reproduce the error your reported. May I ask you to check the following:
- [ ] Do you have only one version of Webots installed on your computer (e.g., the one compiled from the `experimental` branch?
- [ ] If not, are you sure the `WEBOTS_HOME` environment variable points to the current installation of Webots
- [ ] Also, the `LD_LIBRARY_PATH` may be a cause of problems if it points to a folder of a different version of Webots.
- [ ] Can also you reproduce the problem with the sample `client` provided in the `player` folder?

medrimonia, 2021-05-12:
I tried reproducing the bug I mentioned, but was not able to do so again. I will therefore close this issue and make sure that if I stumble upon it again I provide a detailed procedure to make sure it is easy to reproduce. I will also make sure to test this with sample client.

closed 2021-05-12 by medrimonia