RoboCup-Humanoid-TC/webots #115: Webots just crashes after running client

SoheilKhatibi, 2021-05-20:
On release branch, the player controller of the robot just crashes after running the `client` or `test_client`.

medrimonia, 2021-05-20:
Dear Soheil, a minimum of additional information would be appreciated. How do you notice that the player controller crashed?
Is there any error message provided?

SoheilKhatibi, 2021-05-20:
There are just 2 warnings about the request from the client side. Does it work okay for you?
Here is the screenshot of what I get:
[image]

medrimonia, 2021-05-20:
Are you certain that you followed the following procedure?

1. `git clean -xfd` _after_ checkout of the release branch
2. `make` in webots folder
3. `make` in `projects/samples/contests/robocup/controllers/player`

This error is reached if the `getNodeType` is not valid which is something that might be impacted if the build is not 'clean' (see #104 )

SoheilKhatibi, 2021-05-20:
After re-building the project the problem is resolved. Thanks. I had built it after checking out to release branch, but maybe after the changes it was needed again.

closed 2021-05-20 by SoheilKhatibi