RoboCup-Humanoid-TC/webots #121: Ubuntu 18.04 support for robot_client was dropped

schra, 2021-05-20:
It's currently not possible to compile `robot_client` under Ubuntu 18.04 because the shipped protobuf doesn't ship `ByteSizeLong`:

```
In file included from /usr/include/arpa/inet.h:22:0,
                 from robot_client.cpp:7:
robot_client.cpp: In member function ‘void RobotClient::sendRequest(const ActuatorRequests&)’:
robot_client.cpp:150:48: error: ‘const class ActuatorRequests’ has no member named ‘ByteSizeLong’; did you mean ‘ByteSize’?
   const uint32_t size = htonl(actuator_request.ByteSizeLong());
                                                ^
robot_client.cpp: In member function ‘void RobotClient::updateHistory(const SensorMeasurements&)’:
robot_client.cpp:238:31: error: ‘const class SensorMeasurements’ has no member named ‘ByteSizeLong’; did you mean ‘ByteSize’?
   uint32_t msg_size = sensors.ByteSizeLong();
                               ^~~~~~~~~~~~
                               ByteSize

```

I solved this by cloning and installing the newest protobuf manually and then everything compiles fine. But I assume you didn't intentionally drop Ubuntu 18.04 support, @medrimonia?

medrimonia, 2021-05-21:
This compilation issue was not raised using 20.04, but yes, that was not intentional. Could you confirm that the fix of #119 solves the problem with default packages of Ubuntu 18.04?

schra, 2021-05-23:
Yes, I think the commit probably works (but I haven't tried it out yet)

closed 2021-05-23 by schra