RoboCup-Humanoid-TC/webots #127: Current implementation does not support feet composed of multiple solids

medrimonia, 2021-05-28:
From the Hamburg Bit-Bots:

> We have a problem with our robot model and the autoreferee. Our feet each have four cleats that touch the ground. For the falling detection, the autoreferee checks whether the solids touching the ground are labeled foot. But we are only allowed to tag a total of four solids with foot and  hand, or else the referee complains with "player is missing a hand or a foot".

Some robots have feet composed of multiple solids and with the current implementation they have to face one of the two following issues:

1. If all solids are flagged as `[foot]`, then the referee exits complaining about the total number of `[hand]+[foot]`, it is not possible to simply set this value to a minimum of 4 because this would slow down the computation of the *ball holding* rule.
2. If only the main solid is flagged as `[foot]`, then the robot is considered as walking with unknown body parts and get penalized for being fallen.

omichel, 2021-05-28:
Changing the structure of the feet is a workaround to this problem: A single solid may contain a complex shape including a compound bounding object. This way the fours cleats may be included in the "[foot]" solid. See https://cyberbotics.com/doc/guide/tutorial-5-compound-solid-and-physics-attributes for an example to implement a compound solid.

SammyRamone, 2021-05-28:
> Changing the structure of the feet is a workaround to this problem: A single solid may contain a complex shape including a compound bounding object. This way the fours cleats may be included in the [...]

The issue is, that we have 4 force sensors in each of the feet, not only solids. Does this also works in this case?

omichel, 2021-05-28:
> The issue is, that we have 4 force sensors in each of the feet, not only solids. Does this also works in this case?

I see. Indeed, my proposal won't work in that case. I assume the force sensors are on each cleat, right?

SammyRamone, 2021-05-28:
> > The issue is, that we have 4 force sensors in each of the feet, not only solids. Does this also works in this case?
> 
> I see. Indeed, my proposal won't work in that case. I assume the force sensors are on each cleat, right?

Yes exactly. Each foot has 4 cleats at the corner and each one has a force sensor.

omichel, 2021-05-28:
In that case, the patch on the referee proposed by @medrimonia is probably the way to go.

medrimonia, 2021-06-03:
Patch was included in the current release branch.

closed 2021-06-03 by medrimonia