RoboCup-Humanoid-TC/webots #137: referee: rotate_along_z does not rotate along z

timonegk, 2021-06-01:
**Describe the Bug**
In the auto-referee, the function `rotate_along_z` currently converts the rotation to a quaternion, multiplies it with the unit quaternion, and converts it back to axis angle. This does not actually rotate along the z axis.
For our robot, that's a problem because it falls down when it is spawned on the other side of the field.

**Screenshots**

https://user-images.githubusercontent.com/25013222/120329702-65beb700-c2ec-11eb-8b71-39031319916f.mp4

omichel, 2021-06-01:
That's correct. For some reason, the rotation quaternion was commented out and the unit quaternion was used instead. I will investigate and fix it as soon as possible. Did you try to uncomment the Z rotation quaternion to see if that resolves your problem?

timonegk, 2021-06-01:
Okay, [0,0,0,1] is not actually the unit quaternion, because the library uses the wxyz order. That actually is a rotation of 180° degrees around the z axis, but the result is wrong because the rotation is not around the global z axis. The commented out line rotates around the x axis, which is also wrong.

timonegk, 2021-06-01:
I thin rotate_along_z should be called before applying the rotation from the startingPose.

timonegk, 2021-06-01:
So that would be keeping rz as [0,0,0,1], but switching the arguments of qmul, that change fixes the issues.

omichel, 2021-06-02:
Fixed in #139.

closed 2021-06-02 by omichel