RoboCup-Humanoid-TC/webots #143: `player` should not disable the actuators when a robot is penalized

timonegk, 2021-06-03:
**Describe the Bug**
When the robot is penalized, it is set to the side of the field and its actuators are disabled. But when the robot is not in a stable position when it is penalized, it cannot reset itself to a stable position and will fall again when the auto referee stops holding it up. The fall results in another penalty and so on.

**Expected behavior**
The robot should be able to move its actuators when it gets penalized so that it can recover from unstable positions.

medrimonia, 2021-06-03:
This issue is forbidding some teams to get back into the game once they are penalized. It should then be solved as soon as possible.

To the best of my understanding, waiting 5 seconds after a penalty before starting to penalize the robot should be sufficient to solve the problem.

omichel, 2021-06-04:
> To the best of my understanding, waiting 5 seconds after a penalty before starting to penalize the robot should be sufficient to solve the problem.

Do you mean waiting 5 seconds after a penalty before starting to disable the actuators of the robot?

medrimonia, 2021-06-04:
Yes, that's exactly what I mean. This would allow robots to set their position properly before disabling the actuators.

omichel, 2021-06-04:
This should be fixed now in #148.

medrimonia, 2021-06-04:
@timonegk Could you confirm that this solves your problem?

omichel, 2021-06-04:
According to https://github.com/RoboCup-Humanoid-TC/webots/pull/148#issuecomment-854718740, I believe this issue can be closed. Feel free to re-open it if needed.

PR #148 (2021-06-04): Added a 5 seconds delay before disabling actuators on penalty
Fixed #143.
closed 2021-06-04 by omichel