RoboCup-Humanoid-TC/webots #150: referee crashes in penalty shootout when robot is spawned upside down

johagge, 2021-06-04:
**Describe the Bug**
We used incorrect poses and spawned the robot upside down. When the robot is spawned and falls headfirst into the ground the referee crashes.
This does not occur when we use our proper poses and is thus probably a low priority issue.

**Steps to Reproduce**
1. set the starting pose for penalty shootout  in the team_1.json to:
      `"shootoutStartingPose": {
        "translation": [2.6, 0, 0.43],
        "rotation": [0, 0.98, 0.13, 0.26]
      },`
2. Use the Wolfgang robot model
3. start webots: `./webots ./projects/samples/contests/robocup/worlds/robocup.wbt`
4. wait until the simulation has loaded and the robot falls
5. See error

<details> 
  <summary>The error messages are the following: </summary>
Waiting for GameController to answer to 240:STATE:SET.
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
Traceback (most recent call last):
  File "referee.py", line 2057, in <module>
    update_ball_holding()  # check for ball holding for field players and goalkeeper
  File "referee.py", line 750, in update_ball_holding
    update_team_ball_holding(red_team)
  File "referee.py", line 700, in update_team_ball_holding
    hull = ConvexHull(points)
  File "qhull.pyx", line 2431, in scipy.spatial.qhull.ConvexHull.__init__
  File "qhull.pyx", line 283, in scipy.spatial.qhull._Qhull.__init__
ValueError: Points cannot contain NaN
java.net.SocketException: Connection reset
	at java.base/java.net.SocketInputStream.read(SocketInputStream.java:186)
	at java.base/java.net.SocketInputStream.read(SocketInputStream.java:140)
	at java.base/sun.nio.cs.StreamDecoder.readBytes(StreamDecoder.java:284)
	at java.base/sun.nio.cs.StreamDecoder.implRead(StreamDecoder.java:326)
	at java.base/sun.nio.cs.StreamDecoder.read(StreamDecoder.java:178)
	at java.base/java.io.InputStreamReader.read(InputStreamReader.java:181)
	at java.base/java.io.BufferedReader.fill(BufferedReader.java:161)
	at java.base/java.io.BufferedReader.readLine(BufferedReader.java:326)
	at java.base/java.io.BufferedReader.readLine(BufferedReader.java:392)
	at controller.net.AutoRefServer.run(Unknown Source)
	at java.base/java.lang.Thread.run(Thread.java:829)
red player 1 has fallen down.
Start listening to AutoReferee connection on port 8750
WARNING: 'referee' controller exited with status: 1.
</details>

**Expected behavior**
The referee should not crash.

**System**
 - Operating System: Linux Ubuntu 18.04
 - Graphics Card: AMD Radeon RX 5600 XT 6GB

**Additional context**
The latest commit of the webots I used was "8a3a920300a6042a1a0a9c084018ad1532a4bebf"

omichel, 2021-06-04:
Shall I assume you are referring to this [Wolfgang robot model](https://github.com/bit-bots/wolfgang_robot/blob/master/wolfgang_webots_sim/protos/Wolfgang.proto)?

johagge, 2021-06-04:
Yes

johagge, 2021-06-04:
I've updated webots to fec4fdeb970113d6b94a23516d6dbb88a837b937.
The changed files in my git are: [150.zip](https://github.com/RoboCup-Humanoid-TC/webots/files/6598838/150.zip) and the Wolfgang robot model linked above (updated to the current version of the master).
Now it happens in the penalty shootout, when its the Darwins turn to do the kick and our robots are set on the side.

I have tested it with 2 teams of Darwins and the issue did not occur.

omichel, 2021-06-04:
Can you try with this modified [referee](https://github.com/RoboCup-Humanoid-TC/webots/files/6599150/referee.zip) controller?
It should fix the crash and ignore the case where the solids seem to fail to return their positions.
Let me know if that helps.

johagge, 2021-06-04:
I did not see a message of the referee crashing.
 
<details> 
  <summary>However the log is now being filled with the following: </summary>
<pre>
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
New state received from GameController: STATE_FINISHED.
Received unexpected state from GameController: STATE_FINISHED while expecting SET
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
</pre>
</details>
The error message is constantly being written to the console, there are now a few hundred lines of just that message.

This happens in the same situtation, when the teams switch and it is the Darwins who are allowed to kick.
What I did notice is that after the minute runs out, the next gamestate is not starting. The State is still "Playing". The time goes to minus values (e.g. -5:19), so I think the referee might actually have crashed.

johagge, 2021-06-04:
The change we describe [here](https://github.com/RoboCup-Humanoid-TC/webots/issues/154#issuecomment-855008692) also fixed this issue for us.

closed 2021-06-04 by johagge