RoboCup-Humanoid-TC/webots #151: referee issues direct_freekick during penalty shootout

SammyRamone, 2021-06-04:
**Describe the Bug**
During penalty kick it can happen that a robot gets ball holding penalty, when the ball is kicked near to it. Then a direct free kick is issued and the penalty shootout sequence gets broken due to this.

**Steps to Reproduce**
1. Start robocup webots world in penalty shootout mode
2. But Ball next to robot
3. Wait till time for the kick is over
3. State is now "DIRECT_FREEKICK"

**Expected behavior**
I'm not sure but I guess that ball holding should be ignored during penalty shootout.

**Video**
https://cloud.mafiasi.de/s/dzbdazc4FPqDHWq

**System**
 - Operating System: Ubuntu 18.04
 - Graphics Card: NVidiea GTX1060

medrimonia, 2021-06-04:
Thanks for pointing that out, I think that the whole aspect of 'fouls' during penalty shootouts need to be addressed, in the rules and in the AutoRef.

medrimonia, 2021-06-15:
According to the vote of June 14th direct free-kick:
- Ball holding does not apply during penalty shootout
- Foul from striker results in automatic end of the trial
- Foul from goalkeeper results in a removal penalty for the rest of the half (to be handled on the GameController side).

medrimonia, 2021-06-22:
Solved as part of #248

closed 2021-06-22 by medrimonia