RoboCup-Humanoid-TC/webots #153: `player` controller dies when a client connects in game state playing

timonegk, 2021-06-04:
**Describe the Bug**
The `player` controller currently dies when our software connects after the game state changed to playing. This is not a problem during the actual games but it is annoying when testing. The `player` also does not seem to terminate correctly because the port is still allocated after the `player` has died.

**Steps to Reproduce**
1. Start the RoboCup environment
2. Wait until the game state changes to playing
3. Connect to the player API

medrimonia, 2021-06-09:
I tried connecting in "PLAYING" state with multiple instances of the default clients and did not experience any issue. Is it still valid with the latest release? If it is, is there any error message that might provide a hint?

SammyRamone, 2021-06-15:
I tested it, it works with the current version of webots. Can be closed in my opinion.

medrimonia, 2021-06-15:
Thank you for the feedback!

closed 2021-06-15 by medrimonia