RoboCup-Humanoid-TC/webots #154: In penalty shootout, joint positions cannot be set

timonegk, 2021-06-04:
**Describe the Bug**
When the game is started in "PENALTY" mode, no joint positions can be set. Adding a debug print at line 409 in player.cpp reveals that the motor positions are actually set correctly by the player, but still, the robot's position in the simulator does not change.

**Steps to Reproduce**
1. Start a game in penalty shootout mode
2. Connect to the player and send motor goals
3. Notice that the robot is not moving

timonegk, 2021-06-04:
This seems to be specific to our robot model, or at least it does not happen with the Darwin

timonegk, 2021-06-04:
In e013edf1c0486103ae8b0419ca3dba334a8d950d, this was still working

SammyRamone, 2021-06-04:
> In [e013edf](https://github.com/RoboCup-Humanoid-TC/webots/commit/e013edf1c0486103ae8b0419ca3dba334a8d950d), this was still working

are you sure? it does not work for me on this version

johagge, 2021-06-04:
Embedding the proto fixed this issue for us.
The change to our proto is here: 
https://github.com/bit-bots/wolfgang_robot/pull/171

medrimonia, 2021-06-05:
One of the element raised by Hamburg Bit-Bots is that this might be related to `WbNode::reset` in the following lines in `WbNode.cpp`:

```
  if (isTemplate() && !mProto->isDeterministic())
    // nonDeterministic procedural PROTO must be regenerated on reset
    setRegenerationRequired(true);
```

I am not sure of the semantics behind `isTemplate` and `isDeterministic`, but this would explain why their change of the robot produced a behavior change. I was also informed that this behavior appeared since 9617edee6 in which `resetPhysics` was replaced by `loadState('__init__')`

omichel, 2021-06-05:
I might have an idea about this problem. It could that for regenerated proto, the referee needs to refresh the list of solid nodes (hands and feet, etc.) after resetting the robot with `loadState('__init__')` because this call will actually cause a regeneration of the protos (and new solid node will replace previous ones) and thus invalidate the handles to the solid nodes of the robot. I will try to fix the referee by resetting the lists of solids after calling `loadState('__init__')`.

omichel, 2021-06-05:
This may explain #150 and especially this message displayed by the referee:
```
Error: wb_supervisor_node_get_position() called with a NULL or invalid 'node' argument.
```

jgueldenstein, 2021-06-06:
Would regenerating the proto not also invalidate all the handles to the devices in the player controller?

omichel, 2021-06-06:
> Would regenerating the proto not also invalidate all the handles to the devices in the player controller?

Not for the motors, because the handles are queried each time the player controller receives a new command.
I am not sure regarding the sensors... Maybe we should clear the sensor map...
Did you experience any problem with that?

jgueldenstein, 2021-06-06:
We made some further tests and can not confirm that it fixes #165. We tested it with two robot models, one that we believe is being regenerated: (1) https://github.com/bit-bots/wolfgang_robot/blob/master/wolfgang_webots_sim/protos/Wolfgang.proto
and one that we believe is not being regenerated: (2) https://github.com/bit-bots/wolfgang_robot/blob/master/wolfgang_webots_sim/protos/WolfgangRobocup.proto

When the robot is penalized and therefore `robot.loadState('__init__')` is called. The robot (1) does still receive images but can not set motor goals anymore. The robot (2) still works fine and can move again and also gets sensor data.

omichel, 2021-06-07:
@jgueldenstein: may I kindly ask you to try again #165? I synced its branch with the develop upstream as I believe a fix was introduced in between that may fix the problem. Meanwhile, I will continue to investigate.

PR #165 (2021-06-07): Fix robot reset
Should fix #154 and #155.
closed 2021-06-07 by medrimonia