RoboCup-Humanoid-TC/webots #156: Ball position is reset when a direct free kick happens

Flova, 2021-06-05:
**Describe the Bug**
The rules say, that `A direct free kick is taken from the place where the offense occurred.

**Steps to Reproduce**
1. Do an offence like ball holding, the ball gets teleported to the middle

**Expected behavior**
The ball should stay at the position.

medrimonia, 2021-06-05:
The comment here is too generic and does not exactly match the rule. The place an offense occurred is not necessarily the location of the ball.
There are multiple tests scenarios available to check different kind of fouls, an easily reproducible way would be to either add a scenario or patch one of those scenarios to add a test on the ball position (e.g. forceful contact).

medrimonia, 2021-06-07:
Ball positions should be checked in:
- `ball_holding/multi_robot_kid`
- `forceful_contact/*`

closed 2021-06-13 by medrimonia