RoboCup-Humanoid-TC/webots #157: Ball holding calculation is wrong.

Flova, 2021-06-05:
**Describe the Bug**
A ball-holding offense occurs nearly every time the robot approaches the ball. This is due to an error in the calculation of the ball-holding.

The rules state:
> More than half of the ball's volume must be outside the convex hull of the player, projected to the ground,for the ball to be considered removable.

Currently, a ball-holding is called if the ball enters the convex hull for 1 second regardless of the overlap (see Video).

This is mainly an issue for larger robots as it's very hard to get the ball inside the convex hull of e.g. the darwin.

**Steps to Reproduce**
1. Launch Webots with the Wolfgang model and the viz for the ball holding
2. Move the robot to its `walk ready` pose.
2. Move the ball similar to the video and watch the logs

**Expected behavior**
Only call a ball-holding if the ball is inside the convex hull with more than 50% of its projected area. Reducing the convex hull by the ball radius could be an approximation.

**Screenshots**
https://user-images.githubusercontent.com/15075613/120889716-cefd3d80-c5fe-11eb-8c82-6de190e2129e.mp4

jgueldenstein, 2021-06-05:
The convex hull is currently calculated by the center points of the solids that are tagged [hand] and [foot]. This is inherently flawed since we can just define our center to be at a totally different location and put an offset on the bounding object and visual model.

I know this is to make it computationally more efficient by using these points and using the offset of the ball size but it can be tricked very easily and unintentionally.

jgueldenstein, 2021-06-05:
In my opinion it would be a better solution to get the boundingObject and transformation to it and calculate the maximum extensions in the x and y axis. This would ensure that the system works more as intended and the ball offset could be removed. This would also speed up the calculation if the ball is inside the convex hull since only the center point being in the polygon needs to be checked

medrimonia, 2021-06-05:
The PR mentioned here adjusts the existing implementation to strongly reduce the margin. It is to be noted that this implementation was agreed on by the TC after discussing this topic with Cyberbotics. If you want to change the way it is implemented I think it would need a more formal description of the approach proposed, with at least pseudo-code explaining the different steps and in the best case a well documented pull request to discuss it and ensure it's not breaking the existing tests.

closed 2021-06-07 by medrimonia