RoboCup-Humanoid-TC/webots #161: Penalty goals are awarded for the wrong team

Flova, 2021-06-05:
**Describe the Bug**
If the team kicks a goal during the penalty shootout it's accounted for in the wrong team. 

**Steps to Reproduce**
* Kick a goal using the robot behavior in a penalty shootout

**Expected behavior**
* The goal should be counted correctly

**Screenshots**
https://streamable.com/fk88t8

Flova, 2021-06-05:
The score in the game controller is correct tho

Flova, 2021-06-05:
[image]

medrimonia, 2021-06-05:
One of the potential explanations is that the `update_team` seems to be only called at end of half and not between penalty shootouts. From my point of view, `update_team` should be called each time `flip_side` is called and not only on specific situations.

I don't now if there is something else to be done about this or not.

PR #166 (2021-06-06): Fixed display of score for penalty goals
Fixes #161.
closed 2021-06-06 by medrimonia