RoboCup-Humanoid-TC/webots #172: Penalty kicker is always robot 1

medrimonia, 2021-06-07:
Current implementation of `is_penalty_kicker(team, id)` does not take into account the fact that other robots of the same team might have received a red card.

The penalty kicker should always be the player with the lowest id of the team.

omichel, 2021-06-07:
What if player number 1 has no red card, but has currently a 30 second penalty removal? Shall we decide that the penalty kicker is number 2 in that case?

medrimonia, 2021-06-08:
The penalty kicker should be chosen after setting state to `SET`. In theory, the GameController should remove the penalty of all robots when entering state `SET` (if it doesn't, that's a bug). So no player should be penalized  when this function is called.

omichel, 2021-06-08:
I believe that #177 now fixes this issue.

medrimonia, 2021-06-08:
Perfect, I will add issues to test them and merge it afterwards since it is not urgent I think.

PR #177 (2021-06-11): Fixed determination of kicker player
Fixes #172.

The kicker player is the first player in the team list (starting from '1' to n) which is not currently penalized (30 second removal penalty) and has no red card.
closed 2021-06-11 by medrimonia