RoboCup-Humanoid-TC/webots #173: Referee can consider that ball is out of play when it should not

medrimonia, 2021-06-07:
The Autoref consider that the ball is out of play in: https://www.twitch.tv/videos/1047768522?t=0h12m39s 

The relevant lines from the referee log are:

```
[0852.023|0065.536] Info: Ball kicked by blue player 4.
[0854.049|0065.712] Info: [0, 0, 0.08] [-0.04986572468561704, -0.014595097520858009, 0.07846001386739071]
[0854.049|0065.712] Info: Ball in play, can be touched by any player (moved by 5.20 cm).
[0944.482|0073.296] Info: Ball touched again by blue player 4.
[0948.346|0073.616] Info: The ball has left the center circle after kick-off.
[1103.686|0086.392] Info: 86392: contact between red player 3 and blue player 4.
[1104.396|0086.440] Info: 86440: contact between red player 3 and blue player 4.
[1109.453|0086.816] Info: 86816: contact between red player 3 and blue player 4.
[1120.182|0087.464] Info: 87464: contact between red player 3 and blue player 4.
[1121.101|0087.520] Info: 87520: contact between red player 3 and blue player 4.
[1121.315|0087.536] Info: 87536: contact between red player 3 and blue player 4.
[1123.824|0087.688] Info: 87688: contact between red player 3 and blue player 4.
[1125.871|0087.808] Info: Blue player 4 committed a forceful contact foul on red player 3 (long collision) outside penalty area.
```

omichel, 2021-06-11:
I don't fully understand what's going on here. It seems the GameController stopped the time whereas it should not.
According to the log, the ball is in play, there is no message saying the opposite.
I checked the `referee.py` for messages displayed when it assign `game.in_play = None` and no such message is displayed in the log here. So, why do you believe it's not in play?

medrimonia, 2021-06-11:
Sorry, it seems I didn't paste until far enough:

```
[1125.871|0087.808] Info: Blue player 4 committed a forceful contact foul on red player 3 (long collision) outside penalty area.
[1125.871|0087.808] Info: Ball not in play, will be kicked by a player from the red team.
[1125.871|0087.808] Info: Indirect free kick awarded to red team.
[1125.871|0087.808] Info: Waiting for secondary state: INDIRECT_FREEKICK:0
[1125.871|0087.808] Info: Sending 11003:INDIRECT_FREEKICK:4 to GameController.
```

medrimonia, 2021-06-11:
And here is the full log in case it might be useful.

[log.txt](https://github.com/RoboCup-Humanoid-TC/webots/files/6639806/log.txt)

medrimonia, 2021-06-11:
I just noticed I misunderstood the `Ball not in play, will be kicked by a player from the red team.`

The `Ball not in play` means that the ball just left play, not that it *was* not in play. The call to Indirect free kick is still wrong here but I spotted why and will push a fix to it soon.

commit b2541fc5 (2021-06-12): Partial fix for Indirect Free Kick

Fixes #173  (but still some improvements required
closed 2021-06-12 by medrimonia