RoboCup-Humanoid-TC/webots #190: AutoRef: format of result reporting

robocup-hl-tc, 2021-06-10:
We are currently working on a service that displays the schedule and the results automatically. For this, we would like to easily extract the final result from the AutoRef logs. They could, for example, be in a format like: 
<teamtag> score[score]:score[score] <teamtag>
where the first score is the score during the game and [score] the one including the penalty shoot-outs.

medrimonia, 2021-06-22:
The version in 31ef09b always prints FINAL SCORE: 0 - 0 (at least in penalty shootouts)

medrimonia, 2021-07-01:
In order to properly report the result in an easy to parse way, the result should also write one line with:
`GAME RESULT: <teamTag> wins (<winComment>)`
or 
`GAME RESULT: DRAW`

with examples of `winComment`:

- Regular time
- Overtime 0-1
- Penalty shootout 2-0
- Extended penalty shootout: ball touched more often: 3-2