RoboCup-Humanoid-TC/webots #199: Automated tests messages include errors

medrimonia, 2021-06-12:
Since the introduction of the `tee` trick in the automated tests (see #188 ), the error messages do now include unwanted output:

An example is the following:
```
[41/44] results/win                                         AL lib: (EE) ALCpulsePlayback_streamStateCallback: Received stream failure!
java.net.SocketException: Connection reset
	at java.base/java.net.SocketInputStream.read(SocketInputStream.java:186)
	at java.base/java.net.SocketInputStream.read(SocketInputStream.java:140)
	at java.base/sun.nio.cs.StreamDecoder.readBytes(StreamDecoder.java:284)
	at java.base/sun.nio.cs.StreamDecoder.implRead(StreamDecoder.java:326)
	at java.base/sun.nio.cs.StreamDecoder.read(StreamDecoder.java:178)
	at java.base/java.io.InputStreamReader.read(InputStreamReader.java:181)
	at java.base/java.io.BufferedReader.fill(BufferedReader.java:161)
	at java.base/java.io.BufferedReader.readLine(BufferedReader.java:326)
	at java.base/java.io.BufferedReader.readLine(BufferedReader.java:392)
	at controller.net.AutoRefServer.run(Unknown Source)
	at java.base/java.lang.Thread.run(Thread.java:829)
PASS 10/10
```

Either we should fix the fact that those error messages appear (kill the GC process before killing the socket), or we should ignore errors in there.

omichel, 2021-06-14:
Before the introduction of the `tee` trick, `stderr` was sent into the log file, which I believe was not a good idea because this kind of error were not very visible. I believe we should fix these errors rather than trying to hide them.

medrimonia, 2021-06-14:
I agree with the general principle, here the errors are linked to a connection reset that is apparently not properly handled by the GC. That should be solved but since it happens only at the end of the game it has low_priority while we have other bugs to solve.

I will edit the title of the issue and first message to make it easier to understand what will need to be done once we have time.