RoboCup-Humanoid-TC/webots #200: Problem with html replay of a game with a particular team

MaikePaetzel, 2021-06-12:
For one particular team in the tournament, we are not able to use the HTML replay of the game. According to the AutoRef, the HTML file has been saved successfully to the hard drive. However, when attempting to replay it, you get an XML Parsing Error in the Browser Console. Here is the link to a tar that contains one of these non-playable HTML files: https://cloud.robocup.org/s/wC3tnkDWaxLqNwb 

This is critical since we need to be able to replay all games during the tournament and currently, this particular team can't advance in their tests as well.

BenjaminDeleze, 2021-06-14:
The problem comes from the fact that one object name contains the character `&`. If not sanitized, this character create a bug during the XML parsing.

I will patch Webots to correctly handle the export of `&`.

In the meantime, if you want to make this simulation work, it is enough to remove this `&` character from the name of the object.

medrimonia, 2021-06-14:
Ok, thank you very much for the swift update.

PR #221 (2021-06-15): Merge develop into release
This PR re-sync the robocup release branch with the cyberbotics develop branch.
It should fix #200.
closed 2021-06-15 by medrimonia