RoboCup-Humanoid-TC/webots #206: Display simulation time in replay

medrimonia, 2021-06-13:
Currently it's hard to make the link between the logs and the replay of the games because there is no common clock. It would be nice to have a small clock showing the simulation time in the display.

A very nice touch would be to have the option of toggling its display (I don't know if that's possible with the html replay).

omichel, 2021-06-15:
Another nice possibility to handle this issue would be to start the recording of the animation at time step 0, so that the slider bar of the resulting animation would display the actual virtual time.

medrimonia, 2021-06-15:
Yes I think that would be even better I agree because this wouldn't be confusing for the audience since it is a feature useful only for teams and developers.

omichel, 2021-06-15:
> 
> 
> Yes I think that would be even better I agree because this wouldn't be confusing for the audience since it is a feature useful only for teams and developers.

OK, in that case, I will rework #212 to take this into account and remove the display of the virtual time.

PR #212 (2021-06-15): Improved display
This PR should fix #206, #171 and #170 by improving the display.
The exiting line is made slightly longer, the font size is very slightly reduced.
A new line is added to be able to display:
- [x] simulation time.
- [x] secondary state.
- [x] secondary time.
- [x] yellow/red cards for each player.
- [x] remaining penalty time for each player.

Here is the result:
[image]

[image]
closed 2021-06-15 by medrimonia