RoboCup-Humanoid-TC/webots #217: Robots are not properly moving during 2nd half

medrimonia, 2021-06-14:
Over the last few games, none of the team seemed able to play with more than one robot (the goalkeeper?) during the second half. According to the private logs of teams, it is expected that the robots were attempting to walk, this might be a problem due to the player disabling the motor command.

medrimonia, 2021-06-14:
Understanding of this situation is based on what happens n [Test37](https://cloud.robocup.org/s/jSX4HYtdqPxSQjG), at 4:14/8:16, the second half starts and:

1. All red robots are still moving and at ~4:15:800 state changes to READY, all those robots are penalized
2. Running `egrep -i '(enab|disab)' log.txt` shows that the  actuators of robots other than 1 are never enabled  again

```
[0526.279|0058.784] Info: Disabling actuators of blue player 3.
[1233.459|0153.328] Info: Disabling actuators of red player 3.
[1233.877|0153.328] Info: Disabling actuators of blue player 2.
[1404.753|0183.384] Info: Enabling actuators of red player 3.
[1404.753|0183.384] Info: Enabling actuators of blue player 3.
[1946.114|0255.648] Info: Disabling actuators of red player 1.
[1946.572|0255.648] Info: Disabling actuators of red player 2.
[1947.022|0255.648] Info: Disabling actuators of red player 3.
[1947.643|0255.648] Info: Disabling actuators of blue player 1.
[2064.239|0285.696] Info: Enabling actuators of red player 1.
[2064.239|0285.696] Info: Enabling actuators of blue player 1.
[2153.185|0300.736] Info: Disabling actuators of blue player 1.
[2154.611|0300.752] Info: Disabling actuators of red player 1.
[2175.860|0305.800] Info: Enabling actuators of red player 1.
[2175.860|0305.800] Info: Enabling actuators of blue player 1.
[2679.798|0386.272] Info: Disabling actuators of red player 1.
```

- [ ] The same change as what has been done for `removal penalty` should be applied. This should solve (1)
- [ ] Find out why actuators of other robots are not enabled again and fix it for (2)

medrimonia, 2021-06-14:
For problem 2 (not being enabled again), what I've seen is the following:

```
            if n > 0:
                player['penalized'] = n
            elif 'penalized' in player and player['penalized'] != REMOVAL_PENALTY_TIMEOUT:
                info(f'Enabling actuators of {color} player {number}.')
                customData.setSFString('')
                del player['penalized']
```

The only other place were `player['penalized']` is set is in `send_team_penalties` where it's safe to assume that we pass only once (otherwise it would output repeated message). The most reasonable explanation is then that:

1. Robot gets penalized with `REMOVAL_PENALTY_TIMEOUT` at the beginning
2. Due to a GC event, the penalty is set to 0 directly
3. In this case `n>0` is never true and we never enter in the `disabled` part

I will try reproducing this and see if it can fix the issue.

yasuohayashibara, 2021-06-14:
I think the problem is the following `setVelocity(0)`.
Our robot uses angle control, but if the following command is executed, it seems to stop moving after that.
After commenting out the following line and executing the webots, the robot is able to side entry.

I am currently repeating the verification, so I apologize if I am wrong.

https://github.com/RoboCup-Humanoid-TC/webots/blob/cc1c4df1396b7bb727e50645ca5815f92ee72019/projects/samples/contests/robocup/controllers/player/player.cpp#L367

PR #220 (2021-06-15): Fix actuators disabling
This PR ensure that a similar procedure is taken for all the cases where the Autoref `reset` the position of players (i.e. changes both, pose and posture). Those cases are:

1. Removal penalty
2. Set position at the beginning of half time
3. Set position at the beginning of a penalty kick (kicker or goalkeeper)

Since 1 was properly working, 2 and 3 have a behavior similar to 1 now.

Finally, a minimal duration of disabling the actuators is used rather than relying on `secs_till_unpenalized` transition which is expected to fix #217 

Current status
- [x] Automated tests [... cut]
commit 12c05919 (2021-06-15): Now disabling actuators for 1 simulated second after every reset_player

Fixes #217

Robots should now be 'stopped in their action' and properly reset at beginning
of second half and for penalty shootouts.

The problem of actuators not being enabled again should be solved as well.
closed 2021-06-15 by medrimonia