RoboCup-Humanoid-TC/webots #232: Corner kick does not go from mode 0 to 1

SammyRamone, 2021-06-16:
**Describe the Bug**
I think there is an error with the corner kick.
It is triggered correctly, game secondaryState and secondaryTeam is set correctly. SecondaryStateMode is set correctly to 0 and ball is placed at correct position. But then it never goes into the mode 1 (placing). Basically the whole game is stuck for ever in this state then.
I just pulled a few minutes before and was on commit 4daf18161281ccb9b727c0bffd05ad67e2b12b9
[log.txt](https://github.com/RoboCup-Humanoid-TC/webots/files/6661811/log.txt)

**Steps to Reproduce**
1. Put ball behind goal line
2. Corner kick is started
3. Never goes from mode 0 to mode 1

SammyRamone, 2021-06-16:
I just tried it one more time (after running a bit of simulation and trying out the goal kick). This time the behavior was similar but then it crashed after a moment. See last lines in this log
[log.txt](https://github.com/RoboCup-Humanoid-TC/webots/files/6661866/log.txt)

SammyRamone, 2021-06-16:
I tried it one more time (without changing the webots version) and this time it actually worked. I don't really get why this is different.

medrimonia, 2021-06-16:
Ok thank you a lot for the update. On the second log we see that phase 1 was reached, but the behavior seems erratic.

We have this kind of scenario in our test and I've never seen this kind of failures, but I think it highlights that we need more debug output.

commit 70b8ad7d (2021-06-23): Fix for game interruptions that were not stepped out

Fixes #261
Fixes #232

Now ball can only be set as `in play` because secondary seconds elapsed in case
the secondary state is `STATE_NORMAL`.
commit 1d773165 (2021-06-23): Another attempt to fix #232 and #261
PR #264 (2021-06-24): Fix for game interruptions that were not stepped out
Fixes #261
Fixes #232

Now ball can only be set as `in play` because secondary seconds elapsed in case
the secondary state is `STATE_NORMAL`.

Test status: All tests passing

EDIT: Does not fix the issues (or at least not all of them).
closed 2021-06-24 by medrimonia