RoboCup-Humanoid-TC/webots #237: Robot unable to move any DOF exiting "penalized" state using player.cpp controller

egordv, 2021-06-17:
**Describe the Bug**
For Vurtual Humanoid League competition, robot penalization is done by the following part of `projects/samples/contests/robocup/controllers/player/player.cpp controller:
`
      if (customData == "")
        actuators_enabled = TRUE;
      else if (customData == "penalized" && actuators_enabled) {
        // penalized robots gets only their actuators disabled so that they become asleep
        stopMotors();
        actuators_enabled = FALSE;
      } else if (customData == "red_card") {  // robots with a red card get both sensors and actuators disabled
        devices_enabled = FALSE;
        for (const auto &entry : sensors)
          enableSensor(entry.first, 0);
        sensors.clear();
      }
`
When there is "penalized" in customData - our robot freezes.
But after referee.py removes the "penalized" state from customData, our robot is still in freezed state (properly receiving motor commands via protobuf but unable to move any DOF).

**Steps to Reproduce**
1. Open projects/samples/contests/robocup/worlds/robocup.wbt in webots (comit 94eeac2b3f756a667791c02da63070d5d16f1be4 or later)
2. Add our robot's proto https://drive.google.com/drive/folders/1AlJz3ark52psp1Crqlb1uj4Ay61uCPz_?usp=sharing (no meshes, but bounding boxes will be visible)
2. Use sample from projects/samples/contests/robocup/controllers/player/test_client.cpp or any other way to control any DOF of the robot (head_yaw etc) via projects/samples/contests/robocup/controllers/player/player.cpp controller
3. Set customData to "penalized" - robot will freeze
4. Clear customData to "" - robot will remain in freezed state unable to move any DOF

omichel, 2021-06-18:
@egordv: how to you control the motion of your robot? By position, velocity or torque?
When the robot is penalized, the target velocity and target torque are reset to 0 to stop its motion.
However, if you control the robot only in position, you should also reset the velocity to its original value once the robot is not penalized any more so that the position control can work again.
Can you confirm that this fixes the problem for you?

omichel, 2021-06-18:
It should be fixed in #238. Can you test it and confirm?

egordv, 2021-06-18:
Thank you. I will do the test right now.

egordv, 2021-06-18:
Ok, my local test are green. Even with only position control enabled (no velocity commands in outgoing protobuf packets) exiting penalized state works as expected, robot is able to move correctly. Thank you for a fix!

omichel, 2021-06-18:
That's great. Thank you for the tests and feedback!

egordv, 2021-06-19:
Hello Olivier! 
My yesterday quick tests were not precise enough. According to tested patch, it's mandatory to set max velocities at least once (at startup, etc). In more strict test case when there is no velocity commands issued at all - resumeMotors fails to resume a robot (it stays freezed again).
I've created a PR with my version of patching this: https://github.com/RoboCup-Humanoid-TC/webots/pull/241, maybe it helps. Using this patch my stricter tests are OK again.

PR #238 (2021-06-21): Fix reset of motor commands after unpenalization
Fixes #237.

The velocity and force/torque were set to 0 when penalizing robots (to freeze them) and were not reset to their original values when un-penalizing robots. Hence robots had to send constantly the velocity and force/torque to be able to move after getting un-penalized, which is not very intuitive when performing position control.

This PR fixes the problem.
closed 2021-06-21 by medrimonia