RoboCup-Humanoid-TC/webots #243: Display issues during free kicks

timonegk, 2021-06-20:
A free kick has three states: 0, where the robots can stop any infringements and stand up, 1, where the robots can walk to their positions, and 2, where the robot can kick.
In the display, however
* state 0 is not indicated at all
* state 1 is shown as state 0
* state 2 is correctly shown as state 2
In addition, the remaining time for state 1 is only shown on the left, but for state 2, the time is also shown on the right. This should probably be unified.
This is rather low priority because it only affects the display, however, during the public streaming at the competition, it might be confusing for spectators when no indication of state 0 is shown.

medrimonia, 2021-06-20:
While I agree that this issue does not impact game play, it makes it really difficult to follow the situation for teams and lead to a difficult understanding of the situation by both teams and developers.

closed 2021-06-21 by omichel

timonegk, 2021-06-22:
@omichel the remaining time in free kick [0] still seems to be missing

PR #249 (2021-06-21): Fix display of secondary state
Fixes #243.