RoboCup-Humanoid-TC/webots #245: Red cards leads to removal and disabling of players of both teams

medrimonia, 2021-06-20:
In [TEST62](https://cloud.robocup.org/s/Wydem3NZn94nDps) at 7m36, Blue 2 receives a warning while it already have a warning and a yellow card which leads to a red card. This leads to red 2 and blue 2 being penalized while only blue 2 should be penalized.

```
[5894.425|0456.680] Info: Sending 57198:CARD:1:2:WARN to GameController.
[5895.545|0456.728] Info: Red player 2 reset to reentryStartingPose: translation (50 12 0.43), rotation (0.0 -0.0 -1.0 -1.57).
[5895.545|0456.728] Info: Disabling actuators of red player 2.
[5895.545|0456.728] Info: sending red player 2 tp [50, 12, 0.43].
[5895.571|0456.728] Info: Blue player 2 reset to reentryStartingPose: translation (50 -12 0.43), rotation (0 0 1 1.571592653589793).
[5895.571|0456.728] Info: Disabling actuators of blue player 2.
[5895.571|0456.728] Info: sending blue player 2 tp [50, -12, 0.43].
```

omichel, 2021-06-21:
That one is difficult to debug, based only of the log files.
From my investigation, it seems the referee receives a positive number of red cards for both the red and the blue player, see [here](https://github.com/RoboCup-Humanoid-TC/webots/blob/cce1204e87fb3b315767d2beaf1cef03e977482c/projects/samples/contests/robocup/controllers/referee/referee.py#L1124-L1125).
However, from the logs, it seems the referee never sends a WARNING, YELLOW or RED card to the red player 2 before that.
So, the bug could be in the GameController sending a wrong number of red cards for player 2.
I am not sure how to further proceed on this issue...

medrimonia, 2021-06-21:
imho, this could come from a problem of computing `index` which would have the same value if: their's a problem with the `team_color` function which is used only at two locations of the code and might provide always the same results if the input is not what we expect.

The way to proceed would be to look for the test_scenario that already includes a red card and maybe add a test to make sure that the robot from the other team does not get moved away from the field, or to print the messages from the GameController, but in this case I think we would see the red card in the UI display.

medrimonia, 2021-06-21:
I tried to reproduce it using the automated tests, however I was unable to obtain a similar behavior. From there, I think that the best thing is to add more logs on the things that could go wrong (even if unlikely) in order to have more information next time we see a similar error.

omichel, 2021-06-22:
As mentioned in #253, I could sporadically reproduce the problem during the `game_phases/kick_off/scenario_5` test and some other tests.
It was difficult to isolate because it was not systematic.
Finally, I found out that it was caused by another GameController running on the same local network and sending bad updates to the referee.
Could that explain the original issue as well?

medrimonia, 2021-06-23:
When using the debug added in #253 I found systematically the error raised by `team_index`. With this, I was able to reproduce the error properly and by adding a simple debug message:

```
        for team_idx in range(2):
            c = game.state.teams[index].team_color
            error(f"Team with idx {team_idx} has color {c} and id {game.state.teams[index].team_number}")
```
I obtained the following results:
```
[0013.871|0000.000] Info: Sending 1:SIDE_LEFT:7 to GameController.
[0013.871|0000.000] Info: Waiting for GameController to answer to 1:SIDE_LEFT:7.
[0014.071|0000.000] Info: Waiting for GameController to answer to 1:SIDE_LEFT:7.
[0014.271|0000.000] Info: Waiting for GameController to answer to 1:SIDE_LEFT:7.
[0014.472|0000.000] Info: New state received from GameController: STATE_INITIAL.
[0014.472|0000.000] Info: State has succesfully changed to INITIAL
[0014.472|0000.000] Info: New secondary state received from GameController: STATE_NORMAL, phase 0.
[0014.473|0000.000] Error: Team with idx 0 has color BLUE and id 4
[0014.473|0000.000] Error: Team with idx 1 has color BLUE and id 4
[0014.474|0000.000] Error: Failed setting initial state: Traceback (most recent call last):
  File "referee.py", line 2470, in <module>
    game_controller_send(f'SIDE_LEFT:{game.side_left}')
  File "referee.py", line 631, in game_controller_send
    game_controller_receive()
  File "referee.py", line 545, in game_controller_receive
    update_penalized()
  File "referee.py", line 1174, in update_penalized
    update_team_penalized(red_team)
  File "referee.py", line 1138, in update_team_penalized
    index = team_index(color)
  File "referee.py", line 430, in team_index
    raise RuntimeError(f'Wrong team number set in team_index(): {id} != {game.state.teams[index].team_number}')
RuntimeError: Wrong team number set in team_index(): 7 != 4
```
For me it makes no doubt yet that what happens is that the GameController can sporadically send a message of a 'temporary state' that should never be published (e.g. when swapping teams). I'll try a hotfix to ignore such messages.

medrimonia, 2021-06-23:
It should now be properly solved.

closed 2021-06-23 by medrimonia

omichel, 2021-06-23:
That looks like a bug in the GameController...

medrimonia, 2021-06-23:
No, that was most likely a comparison of string or something else and int -> see cd9cdf2

I was getting the sanity check trigerred systematically before and not at all afterwards

omichel, 2021-06-23:
I don't understand. I added this line in the code to display the data types:
```python
    print(f'TYPE = {type(game.state.teams[0].team_number)} {type(id)}.')
```
And I am getting:
```
TYPE = <class 'int'> <class 'int'>.
```
Which mean they are identical and we don't need any case here.
I checked all the `game.json` files I have and they all have the `id` specified as an int:
```json
    "id": 8,
```
Could it be that in your case, it was specified otherwise, like:
```json
  "id": "8",
```
That would explain the problem.
In that case, we should simply ensure after initialization that `game.red.id` and `game.blue.id` are `int` and not `str`.

medrimonia, 2021-06-23:
Thank you for the detailed investigation, you are right about this, and the way to fix it properly. The program generating the game.json was outputing string instead of ints. I would suggest to add a double safety layer:

1. As proposed, ensure that game.red.id is an int in the referee.py
2. Fix the program that generates the game.json to have store int rather than str.

omichel, 2021-06-23:
All right.
Also note `id` generated as `str` are likely causing other bugs in the current version, like https://github.com/RoboCup-Humanoid-TC/webots/blob/cd9cdf2fa4b23a7e4f7f41ee2d7948ec4df0980e/projects/samples/contests/robocup/controllers/referee/referee.py#L386
This is why it should be fixed after the initialization of the game object.
I will take case of this.

medrimonia, 2021-06-23:
Yes, I do fully agree that it can cause a lot of problems and discrepancies with the tests scenarios.

I already fixed this in the generation of the team.json. Thanks for taking care of this in the referee!

commit cd9cdf2f (2021-06-23): Fixing `team_index` + removing useless filtering: Fixes #245
PR #256 (2021-06-23): Fixed team id data type
Fix #245 in a more robust way.