RoboCup-Humanoid-TC/webots #260: Referee and Gamecontroller are out of sync for the kickoff

Flova, 2021-06-23:
**Describe the Bug**
During the second half, the game controller (and therefore the robots) thinks that another team has kickoff as the referee. This leads to the behavior that the teams correctly position themselves, but the robots of the kickoff team are removed by the game controller. It only happens in the beginning second half time and I suspect that the referee is confused due to goals scored in the game and the kickoffs related to the goals.

**Steps to Reproduce**
* Simulate a game with goals
* The kickoff in the second half is wrong in the referee ui vs the gamecontroller ui and this isn't limited to the ui.
 
E.g. second half of https://www.twitch.tv/videos/1065150793

**Expected behavior**
They should be synced

timonegk, 2021-06-23:
This also happens when no goal was scored, and the referee is right, the GameController is wrong.

SGSSGene, 2021-06-24:
The same happend in game Starkit vs RFC Berlin
The kick of side was not switched (our robots didn't receive any signal).

https://cloud.robocup.org/s/fTW228MqMMrS3fN

timonegk, 2021-06-24:
If it helps, here is a log of a game that I just simulated where this issue occurred: https://games.bit-bots.de/private/issue-260/game.html
The referee log is here: https://games.bit-bots.de/private/issue-260/referee_log.txt
According to the game controller, the red team has kick off both times, which is wrong.