RoboCup-Humanoid-TC/webots #261: Direct Freekick without team happens and freezes autoreferee

timonegk, 2021-06-23:
**Describe the Bug**
Sometimes, when a free kick is granted, there is no team associated with the free kick (at least in the display). Then, the secondary state time in the game controller stops running and the auto referee no longer does anything. Logs and videos are attached. The issue now occurred three consecutive times while simulating test games.

**Screenshots**
First game: https://streamable.com/vvc6es
Log: [referee_log.1.txt](https://github.com/RoboCup-Humanoid-TC/webots/files/6703839/referee_log.1.txt)

Second game: (I don't have a video)
Log: [referee_log.2.txt](https://github.com/RoboCup-Humanoid-TC/webots/files/6703855/referee_log.2.txt)

Third game: https://streamable.com/kihjnm
Log: [referee_log.3.txt](https://github.com/RoboCup-Humanoid-TC/webots/files/6703859/referee_log.3.txt)

(To finish the game, I manually changed the state to Finished, this is not done automatically by the referee.)

timonegk, 2021-06-23:
In the GameController GUI, the free kick is shown for one of the teams.
[image]

timonegk, 2021-06-23:
Also, the auto referee just moved two robots away from the ball for the free kick, but they were placed in such a way that they are colliding / spawned inside each other.

medrimonia, 2021-06-23:
Thanks to your log, I feel that I've found something which should not happen:
```
[2428.824|0459.808] Info: Sending 57553:DIRECT_FREEKICK:7 to GameController.
[2428.824|0459.808] Info: Waiting for GameController to answer to 57553:DIRECT_FREEKICK:7.
[2429.024|0459.808] Info: Waiting for GameController to answer to 57553:DIRECT_FREEKICK:7.
[2429.238|0459.808] Info: Ball in play, can be touched by any player (10 seconds elapsed).
[2429.270|0459.808] Info: New secondary state received from GameController: STATE_DIRECT_FREEKICK, phase 0.
```
We see that we have `Ball in play` (which set `game.interruption_team` to `None`) after sending the `FREE_KICK` but before having the `New secondary state`.

commit 70b8ad7d (2021-06-23): Fix for game interruptions that were not stepped out

Fixes #261
Fixes #232

Now ball can only be set as `in play` because secondary seconds elapsed in case
the secondary state is `STATE_NORMAL`.
PR #264 (2021-06-24): Fix for game interruptions that were not stepped out
Fixes #261
Fixes #232

Now ball can only be set as `in play` because secondary seconds elapsed in case
the secondary state is `STATE_NORMAL`.

Test status: All tests passing

EDIT: Does not fix the issues (or at least not all of them).
closed 2021-06-24 by medrimonia